ZSWatch/Dock-HW #2: Question about STM32_TxD and STM32_RxD in Reset-Config

leezhenghui, 2023-10-07:
Look into the schematic design, I am trying to understand the purpose of getting STM32_T(R)xD in the ZSWatch-Reset-Config section of schematic diagram, but did not get an idea. Any clues about this is really appreciated.

Kampi, 2023-10-07:
Hi @leezhenghui,

the selection switch is a debug placeholder for some additional debug purposes. One possible scenario is to configure the reset pin of the watch as UART Rx pin, UART Tx pin, or GPIO in case RTT isn´t working as expected or for other scenarios. They don´t have a function yet.

leezhenghui, 2023-10-07:
> Hi @leezhenghui,
> 
> the selection switch is a debug placeholder for some additional debug purposes. One possible scenario is to configure the reset pin of the watch as UART Rx pin, UART Tx pin, or GPIO in case RTT isn´t working as expected or for other scenarios. The STM32 is working as a VCOM in this scenario (please see `J-Link-OB-STM32F072` in `docs`). But this function isn´t added to the watch firmware for now.

Hi, @Kampi , thanks for your explanation, especially, with a meaningful usage scenario. the design makes sense to me now. :-)

closed 2023-10-07 by leezhenghui