ZSWatch/Dock-HW #27: Add interface for current measurment

Kampi, 2024-07-23:
Add a 2-pin jumper for a current measurement of the watch. A PPK2 can use this interface.

jakkra, 2024-07-24:
Question is if this would show correct readings. Charger can be disabled in SW, but vbus will turn on USB switch. But may still be good for hunting stuff that consumes more than a few uA

Kampi, 2024-07-24:
> Question is if this would show correct readings. Charger can be disabled in SW, but vbus will turn on USB switch. But may still be good for hunting stuff that consumes more than a few uA

That´s true. The USB switch will consume the same amount of power all the time so you have a small offset. IMO it can make sense because otherwise you have to open the watch again and again to measure directly at the battery connector.

jakkra, 2024-07-24:
Yes I agree it still makes sense and is useful

PR #29 (2024-09-13): Release 4
Closes #27 
Closes #30
closed 2024-09-13 by jakkra